sniekum/ar_track_alvar #43: Is X and Y pose data able to be supplied with non kinetic (usb_cam) feed?

mjstn, 2015-02-22:
First of all thank you for your amazing software.  I wish to use it fed from usb_cam and a webcam.

In using usb_cam from a creative webcam I am able to see correct AR tag numbers on topic /ar_pose_marker.    Of course the webcam has no depth so I would not expect to see Z position but I wonder if it is also true that I see x and y always as 0.0 even if I move the patterns?

I can see multiple tags in one field found nicely by ID on /ar_pose_marker topic.    Also of note the orientation even if I rotate the tag is always x:0 y:0 z:0 w: 1.0 but yet it always can report correct tag number so it does find the tags are at least present.

Platform: Beaglebone Black (32 bit) on Ubuntu 14.04.1 running ROS indigo version 1.11.9
Package: ros-indigo-ar-track-alvar
Status: install ok installed
Priority: extra
Section: misc
Installed-Size: 1613
Maintainer: Scott Niekum <sniekum@cs.umass.edu>
Architecture: armhf
Version: 0.5.0-0trusty-20141204-2220-+0000

Launch file (simplified for just usb_cam and ar_track_alvar nodes:
<launch>

   <arg name="marker_size" default="6.6" />
   <arg name="max_new_marker_error" default="0.18" />
   <arg name="max_track_error" default="0.15" />
   <arg name="cam_image_topic" default="/camera/rgb/image_color" />
   <arg name="cam_info_topic" default="/camera/rgb/camera_info" />
   <arg name="output_frame" default="/camera_link" />

   <arg name="video_device" default="/dev/video0" />
   <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" clear_params="true" output="screen">
        <remap from="usb_cam/image_raw" to="/camera/rgb/image_color" />
        <remap from="usb_cam/camera_info" to="/camera/rgb/camera_info" />

        <param name="video_device" value="$(arg video_device)" />
        <param name="image_width" value="640" />
        <param name="image_height" value="480" />
        <param name="framerate" value="5" />
        <param name="pixel_format" value="mjpeg" />
        <param name="contrast" value="120" />
        <param name="brightness" value="180" />
        <param name="saturation" value="32" />
        <param name="camera_frame_id" value="camera_link" />
  </node>

        <node name="ar_track_alvar" pkg="ar_track_alvar" type="individualMarkersNoKinect" respawn="false" output="screen" args="$(arg marker_size) $(arg max_new_marker_error) $(arg max_track_error) $(arg cam_image_topic) $(arg cam_info_topic) $(arg output_frame)" />

</launch>

mjstn, 2015-02-22:
The launch file was cutoff for most of usb_cam part so I try again here:
<launch>
   <arg name="marker_size" default="6.6" />
   <arg name="max_new_marker_error" default="0.18" />
   <arg name="max_track_error" default="0.15" />
   <arg name="cam_image_topic" default="/camera/rgb/image_color" />
   <arg name="cam_info_topic" default="/camera/rgb/camera_info" />
   <arg name="output_frame" default="/camera_link" />

 <arg name="video_device" default="/dev/video0" />
   <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" clear_params="true" output="screen">
        <remap from="usb_cam/image_raw" to="/camera/rgb/image_color" />
        <remap from="usb_cam/camera_info" to="/camera/rgb/camera_info" />

The rest is as shown in 1st post

mjstn, 2015-02-22:
unable to post launch file first part but perhaps comment on if usb_cam non-depth cam has any hope of providing x and y data for the AR tags that do get seen.  Thanks

sniekum, 2015-02-22:
Actually, ar_track_alvar uses the size/distortion of the tag in the image
to calculate depth of the tag, even without a kinect.  So you should be
getting a fully 6-DOF pose estimate.  Hard to say what is wrong though from
the limited info I have about your situation.  Sounds like something might
be wrong in the TF tree connecting your camera frame and output frame.

--Scott

mjstn, 2015-02-23:
Ok and it's good to know the intent is there for webcams to be ok as I need to use webcam and not my Xtion which is way too big and too power hungry for my small device.

I should note that I have had a reply from Patrick Goebel who has recently re-written his 1st book 'ROS By Example Volume 1' for ubuntu 14.04 and ROS Indigo and he also sees null values for AR patterns he knows work with Kinetic.   Patrick is extremely experienced, well beyond my level so it was he that suggested I pass this by you for your input.

From what you then say, I feel the issue is perhaps related to the newer usb_cam node if you yourself know that the ar_track_alvar does indeed work (today) using the no kinect mode for webcam and the ROS usb_cam node to a webcam.

We will dig in farther and perhaps back down to the older uvccam sort of node approach.
Thanks,
Mark Johnston

pirobot, 2015-02-23:
Hello @sniekum and @mjstn.  I am experiencing the same issue as @mjstn under both ROS Hydro (Ubuntu 12.04 32-bit) and ROS Indigo (Ubuntu 14.04 64-bit).  I have tried both the uvc_cam driver by Eric Perko and the usb_cam driver by Bosch.  Both drivers connect to my Logitech Fusion webcam fine and publish an image that I can view using image_view.  I have also verified that the drivers are publishing messages on the camera_info topic including the camera frame (which I have set to camera_link in the launch files).  Here is a typical message on the /camera/rgb/camera_info topic:

	header: 
	  seq: 11591
	  stamp: 
	    secs: 1424707014
	    nsecs: 161616516
	  frame_id: camera_link
	height: 480
	width: 640
	distortion_model: ''
	D: []
	K: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
	R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
	P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
	binning_x: 0
	binning_y: 0
	roi: 
	  x_offset: 0
	  y_offset: 0
	  height: 0
	  width: 0
	  do_rectify: False

My ar_indiv_no_kinect.launch file looks like this:

	<launch>
		<arg name="marker_size" default="20.0" />
		<arg name="max_new_marker_error" default="0.08" />
		<arg name="max_track_error" default="0.05" />
		<arg name="cam_image_topic" default="/camera/rgb/image_raw" />
		<arg name="cam_info_topic" default="/camera/rgb/camera_info" />	
		<arg name="output_frame" default="camera_link" />
	
		<node name="ar_track_alvar" pkg="ar_track_alvar" type="individualMarkersNoKinect" respawn="false" output="screen" args="$(arg marker_size) $(arg max_new_marker_error) $(arg max_track_error) $(arg cam_image_topic) $(arg cam_info_topic) $(arg output_frame)" />
	
	</launch>

And I am using an AR marker that measures 20cm on a side.  This marker is detected fine when using an Xtion Pro camera and the ar_indiv_kinect.launch file with the same parameters as above.  However, when using the Logitech webcam and the launch file above, no marker IDs are echoed in the terminal window used to run the launch file and when I echo the /ar_pose_marker topic I get messages like these:

	header: 
	  seq: 11999
	  stamp: 
	    secs: 0
	    nsecs: 0
	  frame_id: ''
	markers: 
	  - 
	    header: 
	      seq: 0
	      stamp: 
	        secs: 1424707122
	        nsecs: 252278100
	      frame_id: camera_link
	    id: 0
	    confidence: 0
	    pose: 
	      header: 
	        seq: 0
	        stamp: 
	          secs: 0
	          nsecs: 0
	        frame_id: ''
	      pose: 
	        position: 
	          x: 0.0
	          y: 0.0
	          z: 0.0
	        orientation: 
	          x: 0.0
	          y: 0.0
	          z: 0.0
	          w: 1.0

I have tried adjusting the max_new_marker_error and max_track_error parameters through their entire range using rqt_reconfigure and no values produced a different result.

Finally, my tf-tree as echoed by view_frames looks like this:

[image]

However, running 'rosrun tf tf_echo /camera_link /ar_marker_0' yields a series of identity transforms like this:

	At time 1424707452.311
	- Translation: [0.000, 0.000, 0.000]
	- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
	            in RPY [0.000, -0.000, 0.000]

Hope that helps!

--patrick

sniekum, 2015-02-23:
OK, since the output frame is 'camera_link', that rules out the possibility
of this being a TF problem.  One thing I can say almost certainly is that
this is not a problem with ar_track_alvar, but with the camera set up.  My
suspicion is that there is a problem being caused by something in the
camera_info, most likely the empty D,K,R,P vectors for rectification. Those
zeroed out distortion parameters are probably getting multiplied by
something somewhere and causing the poses of the tags to get zeroed out as
well.

Otherwise, you might have to dig into the ar_track_alvar code itself and
print out the pose estimates in image coordinates, then in 3d coordinates
pre-rectification and post-rectification to get s sense for what is going
on.  Good luck!

--Scott

pirobot, 2015-02-24:
Thanks Scott.  That makes sense.  Mark, you might have to [calibrate your camera](http://docs.opencv.org/doc/tutorials/calib3d/camera_calibration/camera_calibration.html) and load the resulting yaml file to make ar_track_alvar work with a webcam.

mjstn, 2015-03-01:
Thank you both for your time.  Have done cal and saved yml files and the values appear now.

closed 2015-03-01 by mjstn