sniekum/ar_track_alvar #46: Jade Release

rctoris, 2015-04-07:
Just a friendly ping for a potential jade release to help some blocking packages for the upcoming beta.

130s, 2015-04-07:
Before making a release into new distro I'd suggest https://github.com/sniekum/ar_track_alvar/issues/42 is taken care of (or at least considered).

sniekum, 2015-04-08:
My ability to keep up with maintaining this package is severely limited right now and is probably only going to get worse in coming months.  Would either/both of you be willing to taking over some of the maintenance duties, since you are some of the more active community members that use the package?  Let me know if you are, and I'll add you as collaborators.

130s, 2015-04-08:
I can join the maintenance to at least review the pull requests and do release work. Running physical tests might depend on the occasions. I think you @sniekum already given me the commit rights.

Since it's already out for Indigo, maybe we want to start out by just releasing from `indigo-devel` branch into Jade (taking back what I said considering sooner release the better)?

sniekum, 2015-04-08:
That's fine with me.

130s, 2015-04-09:
Making a progress. Found that `ar_track_alvar_msgs` is in a separate repository. 

@sniekum can you also add me a write access there?

@wjwwood can you also give me a write access to https://github.com/ros-gbp/ar_track_alvar_msgs-release/ ?

Thanks in advance.

wjwwood, 2015-04-09:
Done.

130s, 2015-04-12:
@sniekum can you also add me a write access at `ar_track_alvar_msgs`? Thx!

sniekum, 2015-04-12:
It looks like you already have it.  Is it not working?

130s, 2015-04-13:
> It looks like you already have it.  Is it not working?

Yeah I see it's working.

130s, 2015-04-14:
[Blocked at resource-retriever](https://github.com/ros/robot_model/issues/104) not being released yet.

130s, 2015-04-24:
Done https://github.com/ros/rosdistro/pull/8150

closed 2015-04-24 by 130s

rctoris, 2015-04-25:
Thanks! :+1: