sniekum/ar_track_alvar #48: Release repository distributed

130s, 2015-04-09:
@jihoonl I'm making the initial release into Jade (related to #46) and I found [the Indigo release repository of this package is under your organization](https://github.com/ros/rosdistro/blob/d8c840ebfdab635940477448aa0ed20af4f8dc83/indigo/distribution.yaml#L194
), instead of [ros-gbp repo used in hydro](https://github.com/ros/rosdistro/blob/6f440516da536f0c69dfca656f4ead19722efa1f/hydro/distribution.yaml#L231) (looks like [this is related](https://github.com/sniekum/ar_track_alvar/issues/27). If there is no specific reason to separate the release repo, does merging those into ros-gbp one make sense (if so I'll take care of it)?

jihoonl, 2015-04-09:
Yeah. I couldn't use ros-gbp because of authorization issue. Please do it in ros-gbp.

130s, 2015-04-14:
Closed via https://github.com/ros/rosdistro/issues/7930

closed 2015-04-14 by 130s