sniekum/ar_track_alvar #55: Confidence is always zero

sudarshan85, 2015-05-07:
Hi,

I'm using ar_track_alvar with a Microsoft webcam on ROS Hydro, Ubuntu 12.04. I am able to detect the tags, however, when I print the ID and the confidence, I get the confidence value to be always zero. The reason I need is that I could set a confidence threshold and reject those detected tags which have a lower confidence value than the threshold. But this can't be done if the confidence value is always zero.
My approach is to do the checking in the code. However, is there any parameters (args) that I can set in the launch file for a confidence threshold?

Thanks.

peter-moran, 2016-07-25:
I am also curious about this, especially for Kalman filtering.

`ar_track_alvar` appears to have a sense of "track error". For example, [MarkerDetector.cpp references track_error](https://github.com/sniekum/ar_track_alvar/blob/f9e9eb89adbd409d9d9a353ff1346e9581dfaedc/src/MarkerDetector.cpp#L111), which is produced by the function [compareCorners()](https://github.com/sniekum/ar_track_alvar/blob/f9e9eb89adbd409d9d9a353ff1346e9581dfaedc/src/Marker.cpp#L163). [The documentation says](http://docs.ros.org/fuerte/api/ar_track_alvar/html/classalvar_1_1Marker.html#ae0f58f40d70f2de959d296200ae0d285) that this is returning "the marker orientation and an error value describing the pixel error relative to the marker diameter."

I think it would make sense to publish this value. My particular use case is that the track_error could be used to improve kalman filtering and help inform a dynamically assigned sensor noise.

130s, 2016-07-25:
Best person who could have an answer to this probably is the original author @sniekum but he might be too busy. And of you want to open a pull request for the proposed change?

peter-moran, 2016-07-27:
With some further investigation, it appears that confidence is used for multi tag bundles, and the confidence is an integer of how many markers in that group. If you want to check, see [`makeMarkerMsgs()`](https://github.com/sniekum/ar_track_alvar/blob/f9e9eb89adbd409d9d9a353ff1346e9581dfaedc/nodes/FindMarkerBundles.cpp#L504) in `FindMarkerBundles.cpp`, which [does assign confidence to a marker](https://github.com/sniekum/ar_track_alvar/blob/f9e9eb89adbd409d9d9a353ff1346e9581dfaedc/nodes/FindMarkerBundles.cpp#L584). Then this [gets called later on](https://github.com/sniekum/ar_track_alvar/blob/f9e9eb89adbd409d9d9a353ff1346e9581dfaedc/nodes/FindMarkerBundles.cpp#L659), passing `bundles_seen[i]` as confidence.

Therefore we should not re-purpose that message, and if you want to access track error we'll need to create another field or message.

peter-moran, 2016-07-28:
I took a look at `track_error` and it does not appear to be useful for determining physical error. Empirically I found a decent linear correlation between the track error and the standard deviation of position and rotation readings while the marker is still, however the whole thing felt a little hacky.

I'm not sure, but `track_error` does not seems to give a good reading on measurement confidence.

peter-moran, 2016-08-02:
Final post. `track_error` is not a good measure of physical confidence. If you look closely at [`makeMarkerMsgs()`](https://github.com/sniekum/ar_track_alvar/blob/f9e9eb89adbd409d9d9a353ff1346e9581dfaedc/nodes/FindMarkerBundles.cpp#L504), it finds the 4 sets of sums, each summing the distance between:
[0] the matching corners, [1] the next corner, [2] the opposite corner, and [3] the previous corner between the old and new blobs. It then finds when the blobs best align (index of minimal value in `errors`) and assigns it as `orientation`. Finally, the resulting error value is the average distance between the corners of the old and new blob, scaled by the size of the tag.

Ultimately, this means that `track_error` will increase with motion and not necessarily true noise. It is not a useful value to consider except for how it is used as "a threshold determining how much tracking error can be observed before an tag is considered to have disappeared."

And, in any case this issue should be closed because confidence does get used. Perhaps it could use better documentation though.

130s, 2016-08-02:
@peter-moran thanks for the deeper look.

Can either of you add a summary to the document? Best location of the document for this package should be http://wiki.ros.org/ar_track_alvar
Note that to edit that wiki page, you'll actually need to edit http://wiki.ros.org/ar_track_alvar/post-fuerte, which is included in the parent page.

closed 2016-08-02 by 130s