sniekum/ar_track_alvar #56: individualMarkersNoKinect with "non stable" camera doesn't work

jnitsch, 2015-05-22:
Hi,

I tried to detect AR tracks with a camera in an arbitrary pose using the node individualMarkersNoKinect. When turning the camera the position of the AR marker changed. When I moved and turned the AR marker, its pose change relative to the camera was alright. But its pose in the world was still wrong. When I put the arm in a position, so that the camera frame had nearly the same orientation like the base frame it seemed to work without any issues.

Is it possible that the transform towards the camera is missing? Or that the camera needs to be aligned the same way like the base frame?

The frames of the camera were published the right way. I am using Rethink Robotics robot Baxter and try to detect the AR track with its hand camera.

Best regards
Julia

sniekum, 2015-05-22:
Can you include your launch file?  Also, what do you mean by "moved" --
with respect to which frame?

jnitsch, 2015-05-22:
This is my launch file:

<launch>
	<arg name="marker_size" default="5.0" />
	<arg name="max_new_marker_error" default="0.1" />
	<arg name="max_track_error" default="0.2" / >
	<arg name="cam_image_topic" default="/cameras/right_hand_camera/image" />
	<arg name="cam_info_topic" default="/cameras/right_hand_camera/camera_info" />	
	<arg name="output_frame" default="/right_hand_camera" />
 
	<node name="ar_track_alvar" pkg="ar_track_alvar" type="individualMarkersNoKinect" respawn="false" output="screen" args="$(arg marker_size) $(arg max_new_marker_error) $(arg max_track_error) $(arg cam_image_topic) $(arg cam_info_topic) $(arg output_frame)" / >
</launch>

about moving the camera: 
I turned Baxter's wrist only. But the camera is not situated in the middle of the wrist, so I changed the camera's pose and so its optical frame with respect to the base frame. 
But I think the parent of the hand camera frame is the wrist frame and this transform doesn't change, I think this is a static one.

sniekum, 2015-05-22:
I need to know what the values of those variable are (especially
output_frame).

jnitsch, 2015-05-22:
Oh I am so sorry! (Copy paste error)

arg name="marker_size" default="5.0" 
arg name="max_new_marker_error" default="0.1" 
arg name="max_track_error" default="0.2" 
arg name="cam_image_topic" default="/cameras/right_hand_camera/image" 
arg name="cam_info_topic" default="/cameras/right_hand_camera/camera_info" 
arg name="output_frame" default="/right_hand_camera"

sniekum, 2015-05-22:
My guess is a TF issue.  I'd switch the output frame to something stable
like Baxter's torso and see if it complains about there not being an
available transform.

jnitsch, 2015-05-22:
I tried it also with the output frame base (also a stable one) and it didn't complain.

closed 2017-04-27 by 130s