sniekum/ar_track_alvar #66: Move include to allow library re-use

timn, 2015-10-09:
We would like to be able to re-use the alvar library in other nodes or software. In particular, we strive to use it as a fallback on Ubuntu for the RoboCup Logistics League Simulation. However, this is currently prevented because the file Camera.h contains the following:
#include <tf/transform_broadcaster.h>

This pulls in a lot of the tf2 infrastructure. This in turn is currently incompatible on our side. The include is actually not used in the library defined by the headers. Rather the only reference is within the nodes. Therefore, if the line was moved out of the header and into the individual nodes, we could use the alvar library and the nodes would work without any difference.

It would be great if this were accepted and could be released as an indigo update. I can provide a patch if desired.

130s, 2015-10-09:
+1 for the pull request ;)

closed 2015-10-12 by sniekum

timn, 2016-02-01:
Hi Scott. Could you release 0.5.2 to jade so we have the fix there as well? It's still 0.5.1 and causing problems for us. Thanks!

sniekum, 2016-02-01:
Due to time constraints, I don’t handle the releases (or maintenance) anymore - there are a few maintainers I’ve asked to help out.  Would someone be willing to do this?  Isaac maybe?

—Scott

130s, 2016-02-01:
Sure.

130s, 2016-02-01:
Done
I https://github.com/ros/rosdistro/pull/10393
J https://github.com/ros/rosdistro/pull/10394

commit 6730630f (2015-10-12): Move tf include from header to cpp files, fixes #66

The header currently prevents us from re-using the library as a given
library (because it pulls in tf2 which causes trouble).

The include has been moved to the individual nodes which actually use a
TransformBroadcaster.
commit 98045beb (2015-10-12): Merge pull request #67 from timn/indigo-devel

Move tf include from header to cpp files, fixes #66