sniekum/ar_track_alvar #73: Activate CI and automate tests

130s, 2016-02-01:
@sniekum Could you activate Travis CI for this repo? I think you can do that at https://travis-ci.org/profile/sniekum with a single click.
Then I'll open a PR for adding travis config.

sniekum, 2016-02-01:
Done.

130s, 2016-02-08:
@sniekum Could you check again https://travis-ci.org/profile/sniekum and see if with `ar_track_alvar` Travis is enabled? Please make sure the package name is correct etc. (I know there's `ar_track_alvar_msgs` package as well).

I've added the config for travis in #74. Travis should run with #74 but it isn't.

130s, 2016-03-23:
Sorry to bother you @sniekum, but could you check again if Travis is activated at:

https://travis-ci.org/profile/sniekum/ar_track_alvar_msgs/settings

and 

https://travis-ci.org/profile/sniekum/ar_track_alvar/settings

130s, 2016-03-23:
> @sniekum Could you check again https://travis-ci.org/profile/sniekum and see if with ar_track_alvar Travis is enabled? 

Sorry correct URLs is https://travis-ci.org/sniekum/

sniekum, 2016-03-23:
Thought I did this a while back, but apparently I didn’t.  Should work now.  Let me know if there are any problems.

—Scott

130s, 2016-03-23:
Yes it's working, thanks!

PR #74 (2016-03-24): Add Travis CI config for indigo branch
Addresses/Fixes #73.
closed 2016-03-23 by 130s