sniekum/ar_track_alvar #91: Kinetic failure on ROS buildfarm prerelease test

130s, 2016-09-27:
ROS prerelease test at the time of https://github.com/sniekum/ar_track_alvar/commit/5c930127804abceab19547b26c871da52acbfcf7 fails only for Kinetic using `kinetic-devel` branch.

```
-- Generating done
-- Build files have been written to: /tmp/catkin_workspace/build_isolated/ar_track_alvar
==> make -j1 in '/tmp/catkin_workspace/build_isolated/ar_track_alvar'
Scanning dependencies of target ar_track_alvar
[  2%] Building CXX object CMakeFiles/ar_track_alvar.dir/src/Camera.cpp.o
In file included from /tmp/catkin_workspace/src/ar_track_alvar/include/ar_track_alvar/Camera.h:47:0,
                 from /tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:25:
/opt/ros/kinetic/include/resource_retriever/retriever.h:81:38: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Retriever(const Retriever & ret) = delete;
                                      ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘bool alvar::ProjPoints::AddPointsUsingChessboard(IplImage*, double, int, int, bool)’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:54:114: error: ‘cvFindChessboardCorners’ was not declared in this scope
  int pattern_was_found = cvFindChessboardCorners(gray, cvSize(etalon_rows, etalon_columns), corners, &point_count);
                                                                                                                  ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:74:72: error: ‘cvDrawChessboardCorners’ was not declared in this scope
                       corners, point_count, false /*pattern_was_found*/);
                                                                        ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::Calibrate(alvar::ProjPoints&)’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:365:29: error: ‘CV_CALIB_USE_INTRINSIC_GUESS’ was not declared in this scope
   &calib_K, &calib_D, 0, 0, CV_CALIB_USE_INTRINSIC_GUESS);
                             ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:365:57: error: ‘cvCalibrateCamera2’ was not declared in this scope
   &calib_K, &calib_D, 0, 0, CV_CALIB_USE_INTRINSIC_GUESS);
                                                         ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::CalcExteriorOrientation(std::vector<CvPoint3D64f>&, std::vector<CvPoint2D64f>&, alvar::Pose*)’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:656:114: error: ‘cvFindExtrinsicCameraParams2’ was not declared in this scope
  cvFindExtrinsicCameraParams2(object_points, image_points, &calib_K, NULL, &ext_rodriques_mat, &ext_translate_mat);
                                                                                                                  ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::CalcExteriorOrientation(std::vector<CvPoint3D64f>&, std::vector<alvar::Point<CvPoint2D64f> >&, CvMat*, CvMat*)’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:692:89: error: ‘cvFindExtrinsicCameraParams2’ was not declared in this scope
  cvFindExtrinsicCameraParams2(&world_mat, &image_mat, &calib_K, &calib_D, rodriques, tra);
                                                                                         ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘bool alvar::Camera::CalcExteriorOrientation(const CvMat*, CvMat*, CvMat*, CvMat*)’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:727:94: error: ‘cvFindExtrinsicCameraParams2’ was not declared in this scope
  cvFindExtrinsicCameraParams2(object_points, image_points, &calib_K, &calib_D, rodriques, tra);
                                                                                              ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::ProjectPoints(std::vector<CvPoint3D64f>&, alvar::Pose*, std::vector<CvPoint2D64f>&) const’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:756:106: error: ‘cvProjectPoints2’ was not declared in this scope
  cvProjectPoints2(object_points, &ext_rodriques_mat, &ext_translate_mat, &calib_K, &calib_D, image_points);
                                                                                                          ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::ProjectPoints(const CvMat*, const CvMat*, const CvMat*, CvMat*) const’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:769:103: error: ‘cvProjectPoints2’ was not declared in this scope
  cvProjectPoints2(object_points, rotation_vector, translation_vector, &calib_K, &calib_D, image_points);
                                                                                                       ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::ProjectPoints(const CvMat*, const alvar::Pose*, CvMat*) const’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:780:106: error: ‘cvProjectPoints2’ was not declared in this scope
  cvProjectPoints2(object_points, &ext_rodriques_mat, &ext_translate_mat, &calib_K, &calib_D, image_points);
                                                                                                          ^
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Homography::Find(const std::vector<alvar::Point<CvPoint2D64f> >&, const std::vector<alvar::Point<CvPoint2D64f> >&)’:
/tmp/catkin_workspace/src/ar_track_alvar/src/Camera.cpp:861:41: error: ‘cvFindHomography’ was not declared in this scope
  cvFindHomography(&src_pts, &dst_pts, &H);
                                         ^
make[2]: *** [CMakeFiles/ar_track_alvar.dir/src/Camera.cpp.o] Error 1
CMakeFiles/ar_track_alvar.dir/build.make:62: recipe for target 'CMakeFiles/ar_track_alvar.dir/src/Camera.cpp.o' failed
make[1]: *** [CMakeFiles/ar_track_alvar.dir/all] Error 2
CMakeFiles/Makefile2:2281: recipe for target 'CMakeFiles/ar_track_alvar.dir/all' failed
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'ar_track_alvar':
  Command '['make', '-j1']' returned non-zero exit status 2

Reproduce this error by running:
Command failed, exiting.
==> cd /tmp/catkin_workspace/build_isolated/ar_track_alvar && make -j1

# END SUBSECTION
```

Also Travis [failed](https://travis-ci.org/sniekum/ar_track_alvar/jobs/162888955) but the cause may seem different.

jimcha21, 2016-11-07:
hello @130s , is there any fix to this ?

achim-k, 2017-01-13:
Was the `kinetic-devel` branch deleted or why do I not see it anymore?

130s, 2017-01-13:
> Was the kinetic-devel branch deleted or why do I not see it anymore?

I'm looking into it now.

130s, 2017-01-13:
>>    Was the kinetic-devel branch deleted or why do I not see it anymore?
>
> I'm looking into it now.

I figured out. I deleted it in https://github.com/sniekum/ar_track_alvar/pull/92, thinking that the branch used in the closed PR was from the requester's forked repo. Turned out that the requester directly committed into the origin somehow (s/he should have a write access to this repo). 
I've undone deletion. Sorry about that.

130s, 2017-01-13:
Back to the original issue of this ticket, it seems fixed in the job that re-ran 20 mins ago. See the [prerelease test](https://travis-ci.org/sniekum/ar_track_alvar/jobs/191794464) in [this build](https://travis-ci.org/sniekum/ar_track_alvar/builds/191794462).
So the cause was either in prerelease test side or somewhere else, not in ar_track_alvar.

closed 2017-01-13 by 130s