sniekum/ar_track_alvar #93: ROS Indigo : version 0.5.4 resulting in erroneous tag detection

unhelkar, 2016-11-15:
[Version 0.5.4](https://github.com/sniekum/ar_track_alvar/commit/1823114d29b008b4cc222f345e29b2e594ec43ca) of ar_track_alvar results in erroneous tag detection when used on ROS Indigo, Ubuntu 14.04 LTS with ASUS Xtion Pro sensor. The tag detection is stable and correct with [0.5.3 version](https://github.com/sniekum/ar_track_alvar/commit/f9e9eb89adbd409d9d9a353ff1346e9581dfaedc), and correct ids are detected upon using `pr2_indiv.launch`. However, incorrect tag values (often very large, 512, 1024 or higher) are detected with 0.5.4.

Note that `sudo apt-get install ros-indigo-ar-track-alvar` currently installs version 0.5.4.

mukeschShah, 2016-11-24:
same problem in my installation, im using  XBOX Kinect360.; ROS indigo, Ubunt 14.04 LTS and the following lauch: 

```
<launch>
	<arg name="marker_size" default="4.4" />
	<arg name="max_new_marker_error" default="0.08" />
	<arg name="max_track_error" default="0.2" />

	<arg name="cam_image_topic" default="/camera/depth_registered/points" />
	<arg name="cam_info_topic" default="/camera/rgb/camera_info" />		
	<arg name="output_frame" default="/camera_rgb_optical_frame" />

	<node name="ar_track_alvar" pkg="ar_track_alvar" type="individualMarkers" respawn="false" output="screen" args="$(arg marker_size) $(arg max_new_marker_error) $(arg max_track_error) $(arg cam_image_topic) $(arg cam_info_topic) $(arg output_frame)" />
</launch>
```

after updating the ID detection is wrong with all markers, before it works well.

unhelkar, 2016-11-24:
@mukeschShah Thanks for confirming the issue. Meanwhile, to have a working copy of the package, one can 
- clone this repository in src folder of catkin workspace (instead of installing via apt-get),
- checkout the commit corresponding to version 0.5.3, and
- catkin_make .

130s, 2016-11-24:
Could any one of you share some `.bag` file(s)? It makes debugging for those who don't always have an access to the environment (camera, tags etc.)

mukeschShah, 2016-11-26:
a .bag file gets very big fast:

https://transferxl.com/08zscZ3wL5R3mf

A **topic** you are particularly interested in?

prarobo, 2016-12-10:
It looks like setting the default marker resolution to 7 in the ros wrapper causes this. If I drop the value back to 5 as in the alvar library, the false detections go away.

mukeschShah, 2016-12-11:
i can confirm prarobo comment
simply add 

```xml
  <arg name="marker_resolution" default="5"
  <node name="ar_track_alvar" pkg="ar_track_alvar" type="individualMarkers" respawn="false" output="screen" args="$(arg marker_size) $(arg max_new_marker_error) $(arg max_track_error) $(arg cam_image_topic) $(arg cam_info_topic) $(arg output_frame) $(arg max_frequency) $(arg marker_resolution) "  />
```
to the launch file and it works again

130s, 2016-12-22:
Anyone interested in sending a fix (via pull request)?

Thanks @mukeschShah for a `.bag` file. Once I get around I'll try integrating that into system testing.

mukeschShah, 2016-12-25:
in my opinion this issue could be closed:
Maybe the documentation and the launch-file should be updated. The documentation gives examples with an `marker_resolution` of 5 and the new default `marker_resolution` in version 0.5.4  is 7. 
The params `marker_resolution` and `max_frequency` should be explained and an example launch-file should be provided.

AndyZe, 2017-01-05:
Why did the default change? That's kind of annoying.

534o, 2017-01-30:
I'm verry sorry for this trouble as @mukeschShah , @prarobo already point out this is my fault at PR #90, so created PR to fix this #104, #105. Please confirm this fixed your problem.

130s, 2017-02-08:
I'll make a new release soon into indigo (for kinetic still waiting for #104 to pass Travis).

@mukeschShah

> Maybe the documentation and the launch-file should be updated. The documentation gives examples with an marker_resolution of 5 and the new default marker_resolution in version 0.5.4 is 7.

Given the popularity of this package, we should maintain backward compatibility. So reverting marker_resolution back to 5 as done in #105 is preferable IMO.

closed 2017-02-08 by 130s

130s, 2017-02-09:
This should be resolved with the following latest released version: 
- Indigo, Jade 0.5.6
- Kinetic 0.6.3

commit f188fa82 (2017-02-08): fix #93, wrong commet at #90 reverted (#105)
commit fea58d18 (2017-02-08): fix #93, wrong commet at #90 reverted (#104)