sniekum/ar_track_alvar #108: [doc] Update individual marker nodes usage about parameter usage

130s, 2017-02-08:
Now that #99 is merged in, [document](http://wiki.ros.org/ar_track_alvar#ar_track_alvar.2BAC8-post-fuerte.Usecase_per_number_of_tags) needs to be updated (otherwise no one uses this nice new feature).

If @achim-k is interested in taking this that'll be great since you're the author of #99 (but no pressure ;) ).

achim-k, 2017-02-08:
I would be interested :wink: Could we include #100 in the new release? Then I can document that additional parameter directly too.

130s, 2017-02-09:
@achim-k Yes now all PRs are merged in, you should be able to update the doc with the changes you contributed. Thanks!

achim-k, 2017-02-09:
Will do so, still waiting for being whitelisted for updating the wiki page. Thanks again!

achim-k, 2017-02-22:
I have troubles finding the right place to edit on the ros wiki. The wiki text editor just shows a few lines like the following
```
{{{#!wiki version groovy_and_newer
<<Include(ar_track_alvar/post-fuerte)>>
}}}
```
and I'm struggling to find the files that are actually included. Could you point me in the right direction? Thanks

130s, 2017-02-22:
@achim-k Yeah you hit the tricky part of ROS wiki. It's here: http://wiki.ros.org/ar_track_alvar/post-fuerte

closed 2017-04-27 by 130s