sniekum/ar_track_alvar #114: ar_track_alvar causes TF_NAN_INPUT

dan9thsense, 2017-02-13:
I am running ar_track_alvar successfully in both ROS indigo and kinetic, but it causes tf transform nodes to output errors such as:

`TF_NAN_INPUT: Ignoring transform for child_frame_id "ar_marker_111"`

I'm pretty sure the errors are irrelevant, as I am detecting all of the expected markers fine and getting good transforms from them. However, these errors make a mess of the terminal screen of other nodes. Is there a way to either suppress the errors or (better) to get ar_track_alvar to only send out valid transforms?

I see that there was a [similar issue](https://github.com/sniekum/ar_track_alvar/issues/35) two years ago without a good resolution.  Any updates since then?

130s, 2017-02-23:
Could it be possible to take a `.bag` file (for only a few seconds is fine) where the issue is occurring and share it?

ZdenekM, 2017-03-08:
Similar issue here. Getting:

```
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "ar_marker_0" from authority "/table/ar_track_alvar_bundle" because of an invalid quaternion in the transform (-nan -nan -nan -nan)
         at line 253 in /tmp/binarydeb/ros-indigo-tf2-0.5.15/src/buffer_core.cpp
```

It's usually quite spurious - hard to capture in bag. Btw, what topics should be recorded? TF and ar_pose_marker or even image?

dan9thsense, 2017-03-08:
It only happens when using depth sensing, not when just using a camera without depth.  The bag file gets huge fast, so as ZdenekM asks, what topics are needed?

130s, 2017-03-08:
For a bag, include only image topic(s) that you want to pass to `ar_track_alvar`. No need for `tf`, or you don't even need to run ar_track_alvar while capturing a bag, since that bag will be the one we want to use for test and debugging the updated code. You can record as long as 5 seconds or even less, as long as you can observe the behavior in question.

dan9thsense, 2017-03-08:
ran all day today, ready to record the bag file and, of course, the error doesn't show.  I'll catch it yet.

closed 2017-04-27 by 130s