sniekum/ar_track_alvar #122: Move to ros-perception org?

130s, 2017-04-26:
@sniekum Occasionally I have had to ask you for admin tasks, and there will still be some cases I/someone have to.
- Do you mind giving me an admin access to this repo?
- Alternatively, even better is to fork this to somewhere OSRF manages, e.g. [ros-perception](https://github.com/ros-perception/) org unit. 
  - Benefit is, you'll be freed from any admin chores, and the package will probably get even more visibility (despite the popular this package has already got). You'll remain the author of course.
  - Drawback is people who downloaded the source will have to adjust their `git remote` to the new repo's location.

sniekum, 2017-04-26:
This is just a normal user repo, so I can’t give you admin access without actually transferring ownership of the repo to you (which is an option we can consider).  I’m also fine with moving this to ros-perception.  What do you think is best?

—Scott

130s, 2017-04-27:
Thanks, I didn't know the limitation of a normal user repo.
Then, I prefer to "fork"ing the repo to `ros-perception` or somewhere under `ros-*` and make that as an official, so that we'll be free from the number of admins limit.

@wjwwood could you fork this repo to a GitHub org unit somewhere appropriate that OSRF maintains, and add @sniekum and @130s as admin?

wjwwood, 2017-04-27:
Yes sir: https://github.com/ros-perception/ar_track_alvar

130s, 2017-04-27:
Thanks @ wjwwood.

> - [ ] Move issues, PRs

Using [github-issues-import](https://github.com/IQAndreas/github-issues-import), after successfully imported 12 issues/PRs only, the import stopped (I might have hit https://github.com/IQAndreas/github-issues-import/issues/41) So I manually imported ONLY open issues 1by1 using https://github-issue-mover.appspot.com/ As a result, only 12 closed issues/merged PRs, all 25 open issues are moved, have been moved to the new repo.

> - [ ] Update dev repo in rosdistro

Done https://github.com/ros/rosdistro/pull/14698

130s, 2017-04-27:
@sniekum hopefully one last thing; could you update at the top page of this repo the summary as something like "MOVED TO https://github.com/ros-perception/ar_track_alvar"?

carlosjoserg, 2017-04-27:
Hi guys, apologies to jump in, it might save you time.

Adding contributors with write access is feasible even for normal users.

Though I believe the best option would be to transfer ownership of the repo, which ill copy issues and PRs plus avoid confusion and it's cleaner. 

Although changing the remotes is recommended, github handles all traffic forwarding very well, I have used it before. Check the last item here: https://help.github.com/articles/about-repository-transfers/

But of course all up to you!

130s, 2017-06-07:
> @sniekum hopefully one last thing; could you update at the top page of this repo the summary as something like "MOVED TO https://github.com/ros-perception/ar_track_alvar"?

@sniekum friendly ping :)
Recently there was still a pull request against this repo.

sniekum, 2017-06-09:
Done.

> On Jun 7, 2017, at 10:05 AM, Isaac I.Y. Saito <notifications@github.com> wrote:
> 
> MOVED TO https://github.com/ros-perception/ar_track_alvar <https://github.com/ros-perception/ar_track_alvar>

130s, 2017-06-12:
Thank you @sniekum!

closed 2017-06-12 by 130s