sniekum/ar_track_alvar #124: properly transfer ar_track_alvar

ahundt, 2017-06-13:
I thought this was the main repo, could you transfer it?
https://help.github.com/articles/transferring-a-repository-owned-by-your-personal-account/

130s, 2017-06-13:
IMO transferring the ownership isn't a requirement when the repository location is changed.
Ref. [MaintenanceGuide#Claiming_Maintainership on ros.org](http://wiki.ros.org/MaintenanceGuide#Claiming_Maintainership) does not mention it, so does not [the recent discussion](https://discourse.ros.org/t/releasing-repositories-form-other-people/1797) about maintainership transfer. Besides, forked repos (now this repo is) can have their own purpose. If you still think the repository should be transfered, could you ask on the linked discourse.ros.org thread?

Meanwhile I'm curious what made you think this repo is the main repo for ar_track_alvar. There's now [a notice at the top page](https://github.com/sniekum/ar_track_alvar/issues/122#issuecomment-306823878), links from [wiki page](http://wiki.ros.org/ar_track_alvar) are updated. If there are still info not updated we need to fix that, so please let us know. Thanks.

ahundt, 2017-06-13:
Yes I saw the notices, however there is still an actual problem my suggestion would fix. Please look at the top of [ros_perception/ar_track_alvar](https://github.com/ros-perception/ar_track_alvar) page. There is a message there:

> forked from `sniekum/ar_track_alvar`

Every single fork points to your repository, that's slightly inconvenient but there is a second problem.

Next, [click this to search "tag" in `ros-perception/ar_track_alvar`](https://github.com/ros-perception/ar_track_alvar/search?q=tag&type=Code&utf8=%E2%9C%93), you will see this error:

> Sorry, forked repositories are not currently searchable. 

As you can see, it is impossible to search in the other repository, because github thinks this is the main repository.

Either transferring or deleting and reuploading without a fork is the only way github will correctly reflect your decision to make https://github.com/ros-perception/ar_track_alvar/ the main repository, and this is the reasons I advise the transfer.

It would be extremely helpful if you could take the steps in my first link to fix those problems! :-)